hansemro/kc1-linux #5: Add framebuffer and LCD panel support to device tree

hansemro, 2021-04-16:
Resources:

- https://github.com/hansemro/kernel_omap_otter-common/blob/p-android-omap-3.4/arch/arm/mach-omap2/board-44xx-tablet-panel.c
- https://github.com/hansemro/kernel_omap_otter-common/blob/p-android-omap-3.4/arch/arm/mach-omap2/board-4430kc1-tablet.c
- https://github.com/hansemro/kf_u-boot/blob/p-ti-u-boot-2014.01/board/ti/omap4_kc1/kc1_panel.c

Hardware details:
- Framebuffer address: `0x9fec4000`
- Framebuffer size: `1200kb`
- LCD Panel:
  - 1024x600
  - 16 bit/pixel
  - PWM backlight

hansemro, 2021-04-17:
`arch/arm/boot/dts/omap4-kc1.dts`:
```
/ {
	...
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		framebuffer_region@9fec4000 {
			no-map;
			reg = <0x9fec4000 (600 * 1024 * 4)>;
		};
	};

	chosen {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		framebuffer00@9fec4000 {
			compatible = "simple-framebuffer";
			reg = <0x9fec4000 (600 * 1024 * 4)>;
			width = <600>;
			height = <1024>;
			stride = <(600 * 4)>;
			format = "r5g6b5";
			status = "okay";
		};
	};
	...
};
```

Kernel log:
```
[    0.842529] simple-framebuffer 9fec4000.framebuffer00: framebuffer at 0x9fec4000, 0x258000 bytes, mapped to 0x(ptrval)
[    0.842590] simple-framebuffer 9fec4000.framebuffer00: format=r5g6b5, mode=600x1024x16, linelength=2400
[    0.850769] Console: switching to colour frame buffer device 75x64
[    0.855682] simple-framebuffer 9fec4000.framebuffer00: fb0: simplefb registered!
```

Before the login prompt, the kernel seems to hang.

hansemro, 2021-04-17:
Adding the following lines to enable dss (which is disabled by default) solved the hanging issue:
```
&dss {
	status = "okay";
};
```

hansemro, 2021-04-17:
Never mind, the device still hangs before the login prompt can be reached.

hansemro, 2021-04-19:
Setting framebuffer address to `0x9e600000` seems to create a framebuffer device without halting the system. However, I can't be certain about the address.

hansemro, 2021-04-19:
```
amazon-otter:~# cat /sys/class/graphics/fb0/phys_addr 
9e800000
amazon-otter:~# cat /sys/class/graphics/fb1/phys_addr 
9f000000
amazon-otter:~# uname -a
Linux amazon-otter 3.4.48+ #24 SMP PREEMPT Tue Mar 2 01:53:11 PST 2021 armv7l GNU/Linux
```

hansemro, 2021-04-20:
Leaving the framebuffer address at 0x9fec4000 and reserving from 9fec0000 onwards, the penguins show up. However, the system still hangs.

[image]

hansemro, 2021-04-20:
[hang_5.11.log](https://github.com/hansemro/kc1-linux/files/6345128/hang_5.11.log)

mj-sakellaropoulos, 2021-04-21:
I can confirm that with the same framebuffer entry, I get the same result.
There is variability in the following:
- Presence of penguins
- Image staying on screen indefinitely
- Reboot after black screen

Constant: Foreground color always yellow.

Right now I'm trying to port the DPI pinmux configuration from uBoot and configure DSS to use DPI LCD2
https://github.com/hansemro/kf_u-boot/blob/p-ti-u-boot-2014.01/board/ti/omap4_kc1/kc1_mux_data.h
https://www.kernel.org/doc/Documentation/arm/OMAP/DSS
https://elixir.bootlin.com/linux/v5.10.31/source/Documentation/devicetree/bindings/display/ti/ti,omap-dss.txt
Note: panel-dpi and similar bindings were removed sometime after kernel 4.x

mj-sakellaropoulos, 2021-04-21:
For simple-framebuffer, I think something similar to https://github.com/drebrez/linux/commit/c2fa556f2cc754c8e2114dbbeaa4e8258de3e16a needs to be done. DSS initializing will overwrite the config from u-boot, breaking the framebuffer (?)
However, the relevant portions have been refactored/renamed/moved somewhere else in mainline - still tracking it down

hansemro, 2021-04-21:
FWIW, I've noticed that when the kernel starts handling pin configuration (pinctrl), the screen goes off.

mj-sakellaropoulos, 2021-04-21:
> FWIW, I've noticed that when the kernel starts handling pin configuration (pinctrl), the screen goes off.

Maybe somehow its resetting the backlight IC over SPI

I have tried a lot of combinations so far. Either I can get a yellow screen from the start or the screen turns off right away. Right now I'm blind until the 1v8 serial adapter arrives, so its pretty difficult.

hansemro, 2021-04-21:
I've tested with 3.3 V BusPirate (for reads only) and it seems to work.

mj-sakellaropoulos, 2021-04-21:
Oh good idea, I have one of those

Now that I think of it, the GPIO47 is probably getting reset to LOW (or whatever means off) - turning off the backlight.
Can that pin be forced via devicetree ?

hansemro, 2021-04-22:
> However, the relevant portions have been refactored/renamed/moved somewhere else in mainline - still tracking it down

5.12-5.13+ will likely move to more generic drivers
https://patchwork.kernel.org/project/linux-omap/cover/20210308134627.13056-1-tony@atomide.com/
https://patchwork.kernel.org/project/linux-omap/cover/20210308133910.12454-1-tony@atomide.com/

hansemro, 2021-04-22:
`arch/arm/boot/dts/omap4-kc1.dts`:
```
/* comment out framebuffer in root */

&dss {
	status = "okay";
};

&dsi2 {
	status = "okay";
	vdd-supply = <&vcxio>;
};
```

<details>
<summary>Kernel dmesg</summary>
<pre>
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.11.0+ (clfs@ds3h) (arm-none-linux-gnueabihf-gcc (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 10.2.1 20201103, GNU ld (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 2.35.1.20201028) #33 SMP Wed Apr 21 21:20:33 PDT 2021
[    0.000000] CPU: ARMv7 Processor [411fc093] revision 3 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Amazon Kindle Fire (first generation)
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 16 MiB at 0x9e800000
[    0.000000] OMAP4: Map 0x9fe00000 to (ptrval) for dram barrier
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1148 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] OMAP4430 ES2.3
[    0.000000] percpu: Embedded 20 pages/cpu s52332 r8192 d21396 u81920
[    0.000000] pcpu-alloc: s52332 r8192 d21396 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
[    0.000000] Kernel command line: console=ttyO2,115200n8 androidboot.wifimac=F0A225D07EFA androidboot.serialno=D01EB0A0147327BN androidboot.hardware=otterx rw root=/dev/mmcblk0p9 vram=24M omapfb.mode=lcd2:1024x600@60 omapfb.debug=y omapfb.vram=0:8M omapdss.def_disp=lcd2 omapdss.debug=y omapfb.test=y rootwait console=ttyO2,115200
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 474248K/522240K available (12288K kernel code, 1151K rwdata, 4004K rodata, 1024K init, 6444K bss, 31608K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Running RCU self tests
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU lockdep checking is enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C: platform modifies aux control register: 0x1e070000 -> 0x3e470000
[    0.000000] L2C: DT/platform modifies aux control register: 0x1e070000 -> 0x3e470000
[    0.000000] L2C-310 erratum 727915 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 6 lines
[    0.000000] L2C-310 cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c4, AUX_CTRL 0x7e470000
[    0.000000] random: get_random_bytes called from start_kernel+0x2b4/0x4e4 with crng_init=0
[    0.000000] OMAP clocksource: 32k_counter at 32768 Hz
[    0.000000] clocksource: 32k_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 58327039986419 ns
[    0.000000] sched_clock: 32 bits at 32kHz, resolution 30517ns, wraps every 65535999984741ns
[    0.001190] TI gptimer clockevent: always-on 32768 Hz at /ocp/interconnect@4a300000/segment@10000/target-module@8000
[    0.004882] Console: colour dummy device 80x30
[    0.004913] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.004943] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.004974] ... MAX_LOCK_DEPTH:          48
[    0.004974] ... MAX_LOCKDEP_KEYS:        8192
[    0.005004] ... CLASSHASH_SIZE:          4096
[... 454 lines cut ...]
[    6.065124] DISPC: dispc_mgr_write_gamma_table: channel 1
[    6.070892] DISPC: dispc_mgr_write_gamma_table: channel 2
[    6.076721] DISPC: fifo(0) threshold (bytes), old 32640/32752, new 32640/32752
[    6.084411] DISPC: fifo(1) threshold (bytes), old 32640/32752, new 32640/32752
[    6.092010] DISPC: fifo(2) threshold (bytes), old 32640/32752, new 32640/32752
[    6.099700] DISPC: fifo(3) threshold (bytes), old 32640/32752, new 32640/32752
[    6.107391] DISPC: fifo(4) threshold (bytes), old 0/128, new 0/128
[    6.107391] omapdss_dispc 58001000.dispc: OMAP DISPC rev 4.0
[    6.119934] DISPC: dispc_runtime_put
[    6.123718] DISPC: dispc_save_context
[    6.123718] DISPC: context saved
[    6.131195] omapdss_dss 58000000.dss: bound 58001000.dispc (ops dispc_component_ops)
[    6.131195] DSI: dsi_runtime_get
[    6.143432] omapdss_dsi 58005000.encoder: OMAP DSI rev 3.0
[    6.149291] DSI: dsi_runtime_put
[    6.152893] omapdss_dss 58000000.dss: bound 58005000.encoder (ops dsi_component_ops)
[    6.167144] DISPC: dispc_runtime_get
[    6.167144] DISPC: dispc_runtime_put
[    6.174957] DISPC: dispc_save_context
[    6.174957] DISPC: context saved
[    6.185638] DSS: dss_save_context
[    6.189208] DSS: context saved
[    6.194366] [drm] Initialized omapdrm 1.0.0 20110917 for omapdrm.0 on minor 0
</pre>
</details>

hansemro, 2021-04-22:
Using r5g6b5 color format removes the yellow background. But now it is stretched vertically.
[image]

hansemro, 2021-04-22:
Using r5g6b5 color format removes the yellow background. But now it is stretched vertically.

mj-sakellaropoulos, 2021-04-22:
Have been trying with the omapfb driver. I can confirm that I have the panel timings working in device tree, but not the video source. 
```dts
[    5.086059] panel-dpi panel: failed to find video source
```

See : 
- https://github.com/mj-sakellaropoulos/kc1_linux_configs/blob/main/omap4-kc1.dts
- panel-dpi.c::panel_dpi_probe_pdata() https://github.com/torvalds/linux/blob/16fc44d6387e260f4932e9248b985837324705d8/drivers/video/fbdev/omap2/omapfb/displays/panel-dpi.c#L161
- dss-of.c::omapdss_of_find_source_for_first_ep()
https://github.com/torvalds/linux/blob/90c911ad7445ccec9936763f05fa5db6a3da53be/drivers/video/fbdev/omap2/omapfb/dss/dss-of.c#L152

hansemro, 2021-04-23:
Using your dts, I also see this:

<details>
<summary>partial dmesg</summary>
<pre>
[    5.799377] pinctrl-single 4a100040.pinmux: pin PIN145 already requested by display; cannot claim for 58000000.dss
[    5.811035] pinctrl-single 4a100040.pinmux: pin-145 (58000000.dss) status -22
[    5.818603] pinctrl-single 4a100040.pinmux: could not request pin 145 (PIN145) from group pinmux_dss_dpi_pins  on device pinctrl-single
[    5.831481] omapdss_dss 58000000.dss: Error applying setting, reverse things back
[    5.839447] omapdss_dss: probe of 58000000.dss failed with error -22
</pre>
</details>

mj-sakellaropoulos, 2021-04-23:
> Using your dts, I also see this:

Strange, I'm not getting those messages

> Also, I think `&gpio45 10` should be `&gpio2 13` (32 + 13 = 45)

Ah that makes sense, I've been trying to figure out the GPIO numbers in the manual for a while

hansemro, 2021-04-23:
> Strange, I'm not getting those messages

They do not show up in kernel console as it boots, but show when I run dmesg.

mj-sakellaropoulos, 2021-04-23:
> > Strange, I'm not getting those messages
> 
> They do not show up in kernel console as it boots, but show when I run dmesg. They might show if the log level increases from whatever I am using currently.

Ah, dmesg will have to wait until the serial adapter arrives. USB gadget doesen't seem to work and i've only just recently got userspace working by integrating initramfs into the kernel image.

> Also, I think &gpio45 10 should be &gpio2 13 (32 + 13 = 45)

I've just tried it and it resolved the "enable-gpio" message

hansemro, 2021-04-23:
> Is USB working for you ?

I can only access USB console when the device is able to reach the login prompt...

hansemro, 2021-04-23:
> "enable-gpio"

You may also need to enable the 3.3 V rail connected to the display with gpio_47 pin.

mj-sakellaropoulos, 2021-04-23:
> > "enable-gpio"
> 
> You may also need to enable the 3.3 V rail connected to the display (gpio_47 pin).

Done. Nothing special happened, just resolved the message.

hansemro, 2021-04-23:
I am not sure if the backlight node needs to be specified, but I am working on it here: #11. With older kernels (and in U-Boot), I can confirm that you can use a tool like devmem to set the registers responsible for PWM backlight (GPTimer10 PWM).

hansemro, 2021-04-23:
Graphics format may be RGB565 according to u-boot source: https://github.com/Hashcode/u-boot/blob/9cc129c3f3ef1dd9bbe0caa1ffa3f75e4ca7753c/board/ti/omap4_kc1/kc1_panel.c#L103-L110

mj-sakellaropoulos, 2021-04-29:
[image]

hansemro, 2021-04-30:
Wow, great progress! I will try as soon as I can.

hansemro, 2021-04-30:
Oddly, I don't have backlight.

hansemro, 2021-04-30:
Currently, I am unable to load simple panel and replicate results.

hansemro, 2021-04-30:
hydis panel is missing from panel-simple.c

hansemro, 2021-04-30:
X has finally landed:
[image]

hansemro, 2021-04-30:
X has landed with confirmed working USB OTG: [image]

mj-sakellaropoulos, 2021-04-30:
> hydis panel is missing 

Oh yeah, totally forgot. Oops

What else was needed?

mj-sakellaropoulos, 2021-04-30:
The reason my DTS might not have worked is because when I was debugging I lost my original dts, so I had to decompile the dtb with glitchy display and I restarted from there. Then when I got it working, I tried to port all of my changes to the in-tree DTS. Maybe I missed a property...

hansemro, 2021-04-30:
Things that broke video output for me:
- enabling dsi1 and dsi2
- configure GPIO 37 pin at its proper offset: no video output; even backlight
- configure GPIO 47 pin at its proper offset: no video output; bottom and top halves differ slightly in brightness
- missing panel timings

Things that did not matter:
- reg property: it actually introduces harmless warnings due to cell size mismatch.

mj-sakellaropoulos, 2021-04-30:
Huh, weird. Ok then I will test with your config and see if I get the same result.

mj-sakellaropoulos, 2021-04-30:
> Clock is still wrong, but it is close enough that it works.

Good enough, within spec

> enabling dsi1 and dsi2 , fix: left disabled

This broke it for me.

hansemro, 2021-04-30:
Weird.

mj-sakellaropoulos, 2021-04-30:
I also did all the other things you mentioned, trying to narrow it down now.

hansemro, 2021-04-30:
I posted my dts/dtsi files in `dtb_dev` branch
https://github.com/hansemro/kc1-linux/commit/99a7da03e258381cc880b25de4a130be3c0cc5a2

hansemro, 2021-04-30:
I forgot to change panel compatible target

mj-sakellaropoulos, 2021-04-30:
Ok - I've tested both now. Must be down to differences in omapdrm across kernel versions.

For Linux 5.12-rc8 : DSI1/2 are needed as "ok"

I've tried your DTS as-is and its no longer working. You're on 5.11 right?

hansemro, 2021-04-30:
> I've tried your DTS as-is and its no longer working. You're on 5.11 right?

Yes

hansemro, 2021-04-30:
The same device tree works in 5.12-rc8, weird.

mj-sakellaropoulos, 2021-04-30:
Oh, ok then. Perhaps I broke the code somewhere if that's the case....

I will use the clean branch and check

Technically my kernel is 5.12-rc8(+), but I didn't see any recent changes

hansemro, 2021-04-30:
[image]

mj-sakellaropoulos, 2021-04-30:
Yeah, the kernel I've been running was on the branch with the printks. I am compiling the clean branch now, I trust it works as you say 😂

hansemro, 2021-04-30:
I just removed vcxio from dsi and it still works...

mj-sakellaropoulos, 2021-04-30:
I've just booted from the clean branch and without DSI, it isn't working, and with DSI it's working. Ultimate confusion...

hansemro, 2021-04-30:
[2021-04-30-09-04_v5.11.0.tar.gz](https://github.com/hansemro/kc1-linux/files/6407143/2021-04-30-09-04_v5.11.0.tar.gz)

hansemro, 2021-04-30:
PLL works on your end, which is really interesting.

mj-sakellaropoulos, 2021-04-30:
> 5.11.0_modules.tar.gz
> 2021-04-30-09-04_v5.11.0.tar.gz

This is working

hansemro, 2021-04-30:
> here it is:
> [CLEAN.tar.gz](https://github.com/hansemro/kc1-linux/files/6407199/CLEAN.tar.gz)

I can confirm your results with your image. DSI-enabled DT works, but not the other.

mj-sakellaropoulos, 2021-04-30:
Very interesting. We have a difference somewhere, I'm not sure what though

hansemro, 2021-04-30:
It looks like it has something to do rootfs/initramfs. Using your dtb, I cannot get video output with my postmarketos rootfs.

mj-sakellaropoulos, 2021-04-30:
hmm, my buildroot has basically nothing in it. I do have libdrm installed, along with the test programs, that's about it

hansemro, 2021-04-30:
Can you test with this old pmOS rootfs: https://www.mediafire.com/file/79vftbedy69w2kf/pmos_rootfs.tar.gz/file

User: user
Password: 1234

Notes: 
Add to /etc/inittab:
```
ttyO2::respawn:/sbin/getty -L ttyO2 115200 vt102
ttyGS0::respawn:/sbin/getty -L ttyGS0 115200 vt102
```

mj-sakellaropoulos, 2021-04-30:
Ok, I will try it

[14 more replies]