hansemro/kc1-linux #15: All the information I have found

mj-sakellaropoulos, 2021-04-20:
I think we are the only two people in the world left working on this. It's a nice, well-built device with fairly beefy OMAP, would surely benefit from an updated linux!

- debian linux, with kernel 3.0.21, working console framebuffer (confirmed on my end), rootfs is dead link now, some custom buggy uBoot : https://forum.xda-developers.com/t/5-4-2013-running-real-linux-on-your-kindle.1677085/
- kernel 4.4 with partial device tree : https://github.com/Hashcode/android_device_amazon_otterx-kernel
extract device tree from zImage : https://github.com/PabloCastellano/extract-dtb
extract kconfig from zImage : binwalk -e deflated_zImage
- panel timings (panel-simple.c ?) : https://github.com/Hashcode/u-boot/blob/p-ti-u-boot-2014.01/board/ti/omap4_kc1/kc1_panel.c
panel port: DSS, on DPI port LCD2 
- USB : Probably the same as the omap4-sdp (mainline) judging by u-boot source code
- wifi : on mmc5 sdio, could not get it working on 3.0.21
- OMAP4430 reference manual : https://www.ti.com/lit/ug/swpu231ap/swpu231ap.pdf

I will update this if I find anything else useful

hansemro, 2021-04-20:
I pleasantly surprised someone other than me would be interested in this device. I don't own too many devices, but I can already tell that this one is special.

hansemro, 2021-04-20:
I have not checked out the 4.4 kernel + dtb. Thanks for pointing that out.

hansemro, 2021-04-20:
Aside from mainline kernel, I have tried 3.0.21, 3.0.31, 3.0.72, 3.4.48 downstream kernels. All of which have framebuffer + LCD backlight + touchscreen support. In addition, I was able to get WIFI working with 3.0.X kernels. I originally started my documentation two years ago (privately) on the downstream kernels, but when I figured out how to boot mainline kernels with device tree blobs, I wanted to share my progress.

I actually do remember trying the debian rootfs from XDA from years ago, but was not heavily involved in that project nor do I even have the rootfs backed up. Even so, there is no point in returning to that specific rootfs since the packages are outdated. In fact, I currently use a postmarketOS rootfs which allows me to install and run the latest tools and packages :).

mj-sakellaropoulos, 2021-04-20:
I haven't managed to get the mainline booting yet with my builroot system, I'm going to try out your defconfig and dts. Also by the way, what voltage is the UART / what adapter are you using?

hansemro, 2021-04-20:
UART setup:
[Logic Level Converter](https://www.sparkfun.com/products/14765)
[Sparkfun FT232RL breakout](https://www.sparkfun.com/products/12731)
Note that a resistor has to be swapped out for the logic level converter (as shown below).
[image]
[image]

mj-sakellaropoulos, 2021-04-20:
I am getting some recognizable text, but its mangled, what are your port settings ?

hansemro, 2021-04-20:
115200n8 at 1V8

mj-sakellaropoulos, 2021-04-20:
Ah, look like im going to need that 1.8v adapter then. Thanks!

hansemro, 2021-04-20:
I mean USB serial works, but UART is still recommended

mj-sakellaropoulos, 2021-04-20:
Yeah, I'm pretty sure the kernel is crashing somewhere, because its not enumerating as an ethernet gadget - or anything at all. UART definitely a must.

hansemro, 2021-04-20:
Not to mention, with UART, you get access to U-Boot console which is something you can't get with USB.

hansemro, 2021-04-20:
If you have not already, recompile Hashcode's 2014 u-boot port with device tree enabled. My [fork](https://github.com/hansemro/kf_u-boot) can be summarized by the following changes:
https://github.com/hansemro/kf_u-boot/commit/9f990c0fd608fe2e449f1fee9f1f596e0cc78ee1
https://github.com/hansemro/kf_u-boot/commit/3415e83bcb963a15ef75527677cc6ac2897a0263

Note: I don't believe it was necessary to change the machine type.

mj-sakellaropoulos, 2021-04-20:
> If you have not already, recompile Hashcode's 2014 u-boot port with device tree enabled.

Huh, according to the forum post, I though devicetree was already supported. Thanks.

Right now i'm using the uboot from the debian project until I can properly backup my mmc.

closed 2021-05-07 by hansemro