hansemro/kc1-linux #16: omap4boot usbboot never finds device

mj-sakellaropoulos, 2021-04-21:
What conditions will allow usbboot ?

usbboot is unable to find the device, it never enumerates with the required VID:PID.
```
waiting for OMAP44xx device...
```

hansemro, 2021-04-21:
That utility only works when the tablet is in usbboot mode (the device appears to be off, but is detected by PC as `Texas Instruments, Inc. OMAP4430`). usbboot mode can be triggered in the following ways:
- via UI option in Hashcode 2014 U-Boot port: navigate to the advanced menu option via power button
- via console command in Hashcode 2014 U-Boot port: just run `kc1_usbboot`
- via fastboot: `fastboot oem idme bootmode 4003; fastboot reboot`
- via adb (recovery): `adb shell idme bootmode 4003; adb shell busybox reboot`
- via pin shorting: described in this old [XDA post](https://forum.xda-developers.com/t/devs-needed-for-twrp-port.1356425/page-3#post-19762674)

`usbboot/omap4boot` usage: described [here](https://forum.xda-developers.com/t/root-wip-root-6-2-1-install-twrp-unbrick-any-rom.1405052/)

hansemro, 2021-04-21:
If your device does not have major hardware issues, you can pretty much use the shorting method to recover from bricks.

hansemro, 2021-04-21:
Relevant XDA threads:
https://forum.xda-developers.com/t/firekit-liveusb-repair-kit-1-1-when-youre-about-to-kill-windows.1430038/
https://forum.xda-developers.com/t/root-wip-root-6-2-1-install-twrp-unbrick-any-rom.1405052/

mj-sakellaropoulos, 2021-04-21:
Oh I knew about idme bootmode 4002 but not 4003 good to know thanks!

I just flashed your fork of u-boot and all is well!

closed 2021-04-21 by mj-sakellaropoulos