hansemro/kc1-linux #27: MMC order inconsistency

hansemro, 2021-04-26:
Problem: eMMC starts up sometimes as mmcblk0, mmclbk1, or even mmcblk2. Because the kernel boot images assume a target location, the kernel can sometimes fail to find root.

The issue is made worse or caused by enabling MMC5 for WLAN.

I have recently read a patch and the discussion of this inconsistent behavior on OMAP mailing list: https://www.spinics.net/lists/linux-omap/msg159229.html

Patch TLDR: set aliases for each mmc device in the device tree to fix their order.

- [ ] Confirm consistent behavior with patch

mj-sakellaropoulos, 2021-04-30:
Seems like aliases are already present in omap4.dtsi :
```dts
...
/ {
	compatible = "ti,omap4430", "ti,omap4";
	interrupt-parent = <&wakeupgen>;
	#address-cells = <1>;
	#size-cells = <1>;
	chosen { };

	aliases {
		i2c0 = &i2c1;
		i2c1 = &i2c2;
		i2c2 = &i2c3;
		i2c3 = &i2c4;
		mmc0 = &mmc1;
		mmc1 = &mmc2;
		mmc2 = &mmc3;
		mmc3 = &mmc4;
		mmc4 = &mmc5;
		serial0 = &uart1;
		serial1 = &uart2;
		serial2 = &uart3;
		serial3 = &uart4;
		rproc0 = &dsp;
		rproc1 = &ipu;
	};
        ...
```

hansemro, 2021-04-30:
They were not present in <5.11.x

mj-sakellaropoulos, 2021-04-30:
I should clarify I experienced the issue _with_ those aliases

hansemro, 2021-04-30:
Well, the issue is that &mmc2 should be aliased to mmc0 for it to be consistent with how things were. So, you would have to adjust root parameter to point to `dev/mmcblk1p9` instead of `/dev/mmcblk0p9`

mj-sakellaropoulos, 2021-04-30:
Oh, ok that makes sense.

hansemro, 2021-05-07:
The aliases seem to fix the mapping of mmc block devices. Closing the issue.

closed 2021-05-07 by hansemro