hansemro/kc1-linux #29: Add touchscreen support to device tree

hansemro, 2021-04-28:
Resources:
- https://github.com/hansemro/kernel_omap_otter-common/blob/84763bbf0a9138cb0ba7b42411cd8d0405c1c684/arch/arm/mach-omap2/board-4430kc1-tablet.c#L592-L596
- https://github.com/hansemro/kernel_omap_otter-common/blob/84763bbf0a9138cb0ba7b42411cd8d0405c1c684/arch/arm/mach-omap2/board-4430kc1-tablet.c#L957-L977
- https://github.com/hansemro/kernel_omap_otter-common/blob/p-android-omap-3.4/drivers/input/touchscreen/ili210x.c
- https://github.com/hansemro/kernel_omap_otter-common/blob/p-android-omap-3.4/drivers/input/touchscreen/ilitek.c
- `Documentation/devicetree/bindings/input/ilitek,ili2xxx.txt`

ILITEK 2107QS001K  touchscreen controller:
- tracks only two fingers.
- i2c-2

hansemro, 2021-04-28:
`arch/arm/boot/dts/omap4-kc1.dts`
```
&i2c2 {
	...
	touchscreen@41 {
		compatible = "ilitek,ilitek2110x";
		reg = <0x4>;
		interrupt-parent = <&gpio2>;
		interrupts = <3 IRQ_TYPE_EDGE_FALLING /* gpio_35 TOUCH_IRQ_1 */
			4 IRQ_TYPE_EDGE_FALLING /* gpio_36 TOUCH_IRQ_2 */
			>;
		reset-gpios = <&gpio1 18 GPIO_ACTIVE_LOW>; /* gpio_18; configuration uncertain */
	};
	...
};
```

mj-sakellaropoulos, 2021-05-01:
Change compatible to "ilitek,ili210x" and the driver will load.

There are some errors though:
```
~ # ls /dev/input/
by-path  event0
~ # dmesg | grep ili
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
[    4.222412] i2c-core: driver [ili210x_i2c] registered
[    8.899566] ili210x_i2c 1-0041: probe
[    8.899566] ili210x_i2c 1-0041: GPIO lookup for consumer reset
[    8.909698] ili210x_i2c 1-0041: using device tree for GPIO lookup
[    9.083251] i2c i2c-1: client [ili210x] registered with bus id 1-0041
[   10.416076] ili210x_i2c 1-0041: ili210x_read_reg failed: -16
[   10.429809] ili210x_i2c 1-0041: Unable to get touch data: -16
[   11.466674] ili210x_i2c 1-0041: ili210x_read_reg failed: -16
[   11.472686] ili210x_i2c 1-0041: Unable to get touch data: -16
```

hansemro, 2021-05-02:
https://github.com/hansemro/kernel_omap_otter-common/blob/84763bbf0a9138cb0ba7b42411cd8d0405c1c684/drivers/input/touchscreen/ilitek.c#L914

Looks like VAUX3 is for touchscreen

hansemro, 2021-05-02:
@mj-sakellaropoulos Can you replace/add GPIO 104 for reset?

```
reset-gpios = <&gpio1 18 GPIO_ACTIVE_LOW
	&gpio3 8 GPIO_ACTIVE_LOW
	>;
```

hansemro, 2021-05-02:
I am not totally convinced that the reset is active low btw. It just happened to be in one of the ilitek examples.

mj-sakellaropoulos, 2021-05-02:
I will try that too. I set VAUX3 to always on but it's still getting disabled
As soon as VAUX3 is disabled we get ilitek errors
```
[   27.658447] g_cdc gadget: acm ttyGS0 serial state 0000
[   27.659027] gs_close: ttyGS0 (22c130d3,70c60c71) done!
[   27.659545] gs_open: start ttyGS0
[   27.659545] gs_open: ttyGS0 (22c130d3,a90e9dc6)
[   27.682830] g_cdc gadget: acm ttyGS0 serial state 0003
[   35.708587] g_cdc gadget: acm ttyGS0 req21.22 v0000 i0002 l0
[   35.709899] g_cdc gadget: acm ttyGS0 req21.22 v0003 i0002 l0
[   35.710388] g_cdc gadget: acm ttyGS0 req21.20 v0000 i0002 l7
[   35.710906] g_cdc gadget: acm ttyGS0 req21.22 v0000 i0002 l0
[   39.517517] VAUX3_6030: disabling
[   39.525329] VDAC: disabling
[   39.528411] VANA: disabling
[   39.529357] vwl1271: disabling
[   40.562103] ili210x_i2c 1-0041: ili210x_read_reg failed: -16
[   40.562194] ili210x_i2c 1-0041: Unable to get touch data: -16
[   41.591918] ili210x_i2c 1-0041: ili210x_read_reg failed: -16
[   41.591918] ili210x_i2c 1-0041: Unable to get touch data: -16
[   42.632110] ili210x_i2c 1-0041: ili210x_read_reg failed: -16
```

hansemro, 2021-05-02:
Can you cat the new input device and get touch data?

mj-sakellaropoulos, 2021-05-02:
Something is stopping the driver probe

mj-sakellaropoulos, 2021-05-02:
Ok, VAUX3 set to 3v0, and reset GPIO is as described above
```
[    6.050262] VAUX3_6030: Bringing 1200000uV into 3000000-3000000uV
[    6.060516] VAUX3_6030: ramp_delay not set
[    6.068267] VAUX3_6030: 3000 mV, enabled
```
Nothing in /dev/input, no errors and ili210 driver is not probing

hansemro, 2021-05-02:
Add printk statements [around here](https://github.com/hansemro/linux/blob/60159e9e7bc7e528c103b6b6d47dfd83af29669c/drivers/input/touchscreen/ili210x.c#L480-L484)

mj-sakellaropoulos, 2021-05-02:
The issue is that the probe function (I used to see [this](https://github.com/hansemro/linux/blob/60159e9e7bc7e528c103b6b6d47dfd83af29669c/drivers/input/touchscreen/ili210x.c#L398) printk) is never executed, which is strange because it was working before

mj-sakellaropoulos, 2021-05-02:
Could have something to do with udev or other userspace issues at this point since there are no errors.

mj-sakellaropoulos, 2021-05-02:
From your 5.11.x kernel config:
```
...
#
# Input device support
#
CONFIG_INPUT=y
# CONFIG_INPUT_LEDS is not set
# CONFIG_INPUT_FF_MEMLESS is not set
# CONFIG_INPUT_SPARSEKMAP is not set
# CONFIG_INPUT_MATRIXKMAP is not set

#
# Userland interfaces
#
CONFIG_INPUT_MOUSEDEV=m
# CONFIG_INPUT_MOUSEDEV_PSAUX is not set
CONFIG_INPUT_MOUSEDEV_SCREEN_X=1024
CONFIG_INPUT_MOUSEDEV_SCREEN_Y=600
# CONFIG_INPUT_JOYDEV is not set
CONFIG_INPUT_EVDEV=m
CONFIG_INPUT_EVBUG=m
...
```

hansemro, 2021-05-02:
I manually add modules to /lib/modules/5.XX.X

mj-sakellaropoulos, 2021-05-02:
OK! It's working now!! Totally missed the modules.tar.gz from before.

Coordinates in X11 seem way off / non-repeatable so far

hansemro, 2021-05-02:
> Is there a datasheet out there for this controller ?

I couldn't find one for the one the kindle uses

mj-sakellaropoulos, 2021-05-02:
Seems like all the datasheets available on the internet are missing protocol / register descriptions

hansemro, 2021-05-02:
Finally got around to test it and seems like a pretty good random number generator to me.

mj-sakellaropoulos, 2021-05-02:
😂🤣
yeah....

hansemro, 2021-05-02:
Here is downstream log btw:

<details>
<summary>3.4.48 (I think)</summary>
<pre>
amazon-otter:/usr/share/X11/xorg.conf.d# xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ilitek_i2c                                id=7    [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max      255
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     2305
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     3969
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max        2
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 4542.860915, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4542.860945, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4542.860945, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 837
Event: time 4542.860976, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1236
Event: time 4542.860976, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4542.860976, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4542.860976, -------------- SYN_REPORT ------------
Event: time 4542.868056, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4542.868056, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4542.868086, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 838
Event: time 4542.868086, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1236
Event: time 4542.868086, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4542.868117, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4542.868117, -------------- SYN_REPORT ------------
Event: time 4542.883680, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4542.883711, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4542.883711, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 842
Event: time 4542.883741, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1235
Event: time 4542.883741, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4542.883741, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4542.883741, -------------- SYN_REPORT ------------
Event: time 4542.891523, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4542.891523, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4542.891523, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 849
Event: time 4542.891554, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1234
Event: time 4542.891554, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4542.891554, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4542.891584, -------------- SYN_REPORT ------------
Event: time
Event: time 4542.907179, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1234
Event: time 4542.907179, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4542.907179, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4542.907210, -------------- SYN_REPORT ------------
Event: time 4542.922713, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4542.922743, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4542.922804, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 875
[... 105 lines cut ...]
Event: time 4545.368093, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4545.368124, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4545.368154, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 951
Event: time 4545.368154, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1141
Event: time 4545.368185, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4545.368185, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4545.368185, -------------- SYN_REPORT ------------
Event: time 4545.375845, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 4545.375845, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 5
Event: time 4545.375875, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 898
Event: time 4545.375875, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1153
Event: time 4545.375875, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 20
Event: time 4545.375875, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4545.375906, -------------- SYN_REPORT ------------
Event: time 4545.391470, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 4545.391470, -------------- SYN_REPORT ------------
^Camazon-otter:/usr/share/X11/xorg.conf.d# xinput
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ilitek_i2c                                id=7    [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ twl6030_pwrbutton                         id=6    [slave  keyboard (3)]
</pre>
</details>

mj-sakellaropoulos, 2021-05-02:
I'm working on getting it working via raw i2c in python. That will take a little while

hansemro, 2021-05-02:
<details>
<summary>3.4.48 dmesg</summary>
<pre>
[    2.388854] ilitek_init
[    2.429412] ilitek_i2c_probe, i2c new style format
[    2.434509] ilitek_i2c_probe, IRQ: 0xC3
[    2.434539] ilitek_i2c_register_device, add i2c device, success
[    2.440826] ilitek_i2c_register_device, client.addr: 0x41
[    2.446563] ilitek_i2c_register_device, client.adapter: 0xDE192CA8
[    2.453124] ilitek_i2c_register_device, client.driver: 0xC085057C
[    2.476501] ilitek_i2c_read_tp_info, firmware version 10.2.4.0
[    2.499877] ilitek_i2c_read_tp_info, protocol version: 1.2
[    2.523437] ilitek_i2c_read_tp_info, max_x: 3968, max_y: 2304, ch_x: 30, ch_y: 18
[    2.531402] ilitek_i2c_read_tp_info, max_tp: 2, max_btn: 0
[    2.537384] input: ilitek_i2c as /devices/platform/omap_i2c.2/i2c-2/2-0041/input/input0
[    2.545989] ilitek_i2c_register_device, register input device, success
[    2.553100] ilitek_i2c_register_device, request irq, success
[    2.559143] ilitek_init, register chrdev(247, 0)
</pre>
</details>

hansemro, 2021-05-02:
On downstream kernels, touch data continues to be polled after the finger is on the screen. In contrast, mainline driver polls data once when the finger is on/off the screen.

mj-sakellaropoulos, 2021-05-03:
Data looks good when obtained manually
```
sudo apk add python3
sudo apk add py3_pip
sudo pip3 install smbus2
```

```py
# Run this script as root
import sys
import time
from smbus2 import SMBus, i2c_msg
# import gpio

I2C_BUS_NUMBER = 1
I2C_CHIP_ADDR = 0x41

ILI210X_DATA_SIZE = 64
ILI210X_CMD_READ_TOUCHDATA = 0x10
ILI210X_REG_PANEL_INFO = 0x20
ILI210X_REG_CALIBRATE = 0xcc

i2c_raw_val = 0

def reset_ili210x():
    print("Not implemented")

def init_i2c():
    return SMBus(I2C_BUS_NUMBER)

def read_touch_data(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_READ_TOUCHDATA])
    read = i2c_msg.read(I2C_CHIP_ADDR, ILI210X_DATA_SIZE)
    bus.i2c_rdwr(write, read)
    return list(read)

def main():
    i2c = init_i2c()
    while(True):
        b = read_touch_data(i2c)
        print(b[0:8])
        time.sleep(0.5)
    i2c.close()

if __name__ == "__main__":
    sys.exit(main())
```

hansemro, 2021-05-03:
```
# Run this script as root
import sys
import time
from smbus2 import SMBus, i2c_msg
# import gpio

I2C_BUS_NUMBER = 1
I2C_CHIP_ADDR = 0x41

ILI210X_DATA_SIZE = 64
ILI210X_CMD_READ_TOUCHDATA = 0x10
ILI210X_CMD_PANEL_INFO = 0x20
ILI210X_CMD_CALIBRATE = 0xcc
ILI210X_CMD_GET_FIRMWARE_VERSION = 0x40
ILI210X_CMD_GET_PROTOCOL_VERSION = 0x42

i2c_raw_val = 0

def reset_ili210x():
    print("Not implemented")

def init_i2c():
    return SMBus(I2C_BUS_NUMBER)

def read_panel_info(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_PANEL_INFO])
    read = i2c_msg.read(I2C_CHIP_ADDR, 6*8)
    bus.i2c_rdwr(write, read)
    return list(read)

def read_firmware_ver(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_GET_FIRMWARE_VERSION])
    read = i2c_msg.read(I2C_CHIP_ADDR, 4*8)
    bus.i2c_rdwr(write, read)
    return list(read)

def read_protocol_ver(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_GET_PROTOCOL_VERSION])
    read = i2c_msg.read(I2C_CHIP_ADDR, 2*8)
    bus.i2c_rdwr(write, read)
    return list(read)

def read_touch_data(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_READ_TOUCHDATA])
    read = i2c_msg.read(I2C_CHIP_ADDR, ILI210X_DATA_SIZE)
    bus.i2c_rdwr(write, read)
    return list(read)

def main():
    i2c = init_i2c()
    res = read_panel_info(i2c)
    fw_ver = read_firmware_ver(i2c)
    prot_ver = read_protocol_ver(i2c)
    print("Max touch point: ", res[6])
    print("Max button number: ", res[7])
    print("y_max: ", res[1]*256 + res[0])
    print("x_max: ", res[3]*256 + res[2])
    print("y_ch: ", res[4])
    print("x_ch: ", res[5])
    print("Firmware version: ", fw_ver[0], ".", fw_ver[1], ".", fw_ver[2], ".", fw_ver[3])
    print("Firmware version: ", fw_ver[0], ".", fw_ver[1], ".", fw_ver[2], ".", fw_ver[3])
    print("Protocol version: ", prot_ver[0], ".", prot_ver[1])
    while(True):
        b = read_touch_data(i2c)
        print(b[0:8])
        time.sleep(0.2)
    i2c.close()

if __name__ == "__main__":
    sys.exit(main())# Run this script as root
import sys
import time
from smbus2 import SMBus, i2c_msg
# import gpio

I2C_BUS_NUMBER = 1
I2C_CHIP_ADDR = 0x41

ILI210X_DATA_SIZE = 64
ILI210X_CMD_READ_TOUCHDATA = 0x10
ILI210X_REG_PANEL_INFO = 0x20
ILI210X_REG_CALIBRATE = 0xcc

i2c_raw_val = 0

def reset_ili210x():
    print("Not implemented")

def init_i2c():
    return SMBus(I2C_BUS_NUMBER)

def read_touch_data(bus):
    write = i2c_msg.write(I2C_CHIP_ADDR, [ILI210X_CMD_READ_TOUCHDATA])
    read = i2c_msg.read(I2C_CHIP_ADDR, ILI210X_DATA_SIZE)
    bus.i2c_rdwr(write, read)
    return list(read)

def main():
    i2c = init_i2c()
    while(True):
        b = read_touch_data(i2c)
        print(b[0:8])
        time.sleep(0.5)
    i2c.close()

if __name__ == "__main__":
```

mj-sakellaropoulos, 2021-05-03:
Update : SDL2 demo reading the raw values from i2c
[image]

The coordinates are Little Endian, but in the driver they are reading them as Big Endian - that's part of the problem

hansemro, 2021-05-03:
With CONFIG_DEBUG_DRIVERS set, I am able to see driver print statements:

```
[    3.166687] bus: 'i2c': add driver ili210x_i2c
[    3.166870] i2c-core: driver [ili210x_i2c] registered
[    3.973571] bus: 'i2c': driver_probe_device: matched device 1-0041 with driver ili210x_i2c
[    3.973571] bus: 'i2c': really_probe: probing driver ili210x_i2c with device 1-0041
[    3.973663] ili210x_i2c 1-0041: no pinctrl handle
[    3.973663] ili210x_i2c 1-0041: probe
[    3.974365] ili210x_i2c 1-0041: GPIO lookup for consumer reset
[    3.974365] ili210x_i2c 1-0041: using device tree for GPIO lookup
[    4.117309] driver: 'ili210x_i2c': driver_bound: bound to device '1-0041'
[    4.117309] ili210x_i2c 1-0041: Dropping the link to 48057000.gpio
[    4.117309] ili210x_i2c 1-0041: Dropping the link to 4a310000.gpio
[    4.117675] ili210x_i2c 1-0041: Dropping the link to 48055000.gpio
[    4.118316] bus: 'i2c': really_probe: bound device 1-0041 to driver ili210x_i2c
[    4.118377] i2c i2c-1: client [ili210x] registered with bus id 1-0041
```

mj-sakellaropoulos, 2021-05-03:
Here's my current progress : https://github.com/mj-sakellaropoulos/linux-omap4-kc1/commit/9b6063fba66faca00f770b7dff7e2b6106fcc021

I changed the coordinates to little endian, now the values seem to really scaled down, but cursor is working mostly right otherwise

BIT(finger) operation was causing most events to be ignored - I had to remove that in the python port too

hansemro, 2021-05-03:
Can confirm those changes fixes polling issue from before with good data for both fingers.

hansemro, 2021-05-03:
>  only problem now is X11 coordinate transform

I wonder if this is because resolution is not specified for ili210x_chip

mj-sakellaropoulos, 2021-05-03:
I've been trying to set it manually by following https://wiki.archlinux.org/title/Calibrating_Touchscreen

Using those formulas, it's slightly better but still scaled down - I'm thinking something else is messing with the resolution

> I wonder if this is because resolution field is not specified for ili210x_chip

That could explain it

mj-sakellaropoulos, 2021-05-03:
Oh yeah, they are assuming symetrical resolution ?!
[image]

hansemro, 2021-05-03:
This seems to fix scaling issue
```
	input_set_abs_params(input, ABS_MT_POSITION_X, 0, 3968, 0, 0);
	input_set_abs_params(input, ABS_MT_POSITION_Y, 0, 2304, 0, 0);
```

hansemro, 2021-05-03:
> command 0x20 panel info is never used, so there is no way they are getting the right values for max x/y

I was going to write a function to handle this, but I am not sure how this will get upstreamed...

mj-sakellaropoulos, 2021-05-03:
> I was going to write a function to handle this

I'm trying my hand at it now, no guarantee of 3968x2304 touch resolution on all devices - I have a feeling this might get modified by the calibration command / different firmwares / etc

mj-sakellaropoulos, 2021-05-04:
Update : It's working , https://github.com/mj-sakellaropoulos/linux-omap4-kc1/commit/b0c4503fadd3a0a3f7569f7bf56a47d3a1c5e8e3

I'm not a C developer, so there might be glaring/dumb mistakes but at least it works 😅

hansemro, 2021-05-04:
> Also : ignore changes to device tree - apparently I still needed DSI = ok on 5.11 branch for some reason...

I actually do need dsi1/2 enabled as well. Something about working a dirty messed things up... So  that's been sorted and I will update the device tree shortly.

hansemro, 2021-05-04:
> Actual "Click" event, can move cursor - but can't click anything

Can you check the events and confirm that it follows the multitouch protocol?
https://www.kernel.org/doc/Documentation/input/multi-touch-protocol.txt

mj-sakellaropoulos, 2021-05-04:
The click isn't working because I commented out the BIT(finger) thing, when that function returns false it signifies a click. It's just that the logic that was there already is invalid.

> Can you check the events and confirm that it follows the multitouch protocol?

Checking now

mj-sakellaropoulos, 2021-05-04:
> Can you check the events and confirm that it follows the multitouch protocol?

I don't think it does follow the protocol , just the events are not triggered in any logical way ? And it's buggy (With BIT(finger) enabled), normally the cursor and events wont track a single touch/drag, but you can glitch it by doing a two-point touch and only releasing one touch. This is intermittent

hansemro, 2021-05-04:
```
static bool ili210x_touchdata_to_coords(const u8 *touchdata, ...
...
{
	if (touchdata[0] & BIT(finger))
		return false;
...
}
```

The if statement is definitely not right. When a finger is detected, it will return false for that finger. This is the opposite of what should happen.

After negating the result, we can track finger 1.
```
if (!(touchdata[0] & BIT(finger)))
		return false;
```

hansemro, 2021-05-04:
> Also, it's useful to compile the ili210x driver as a module, which can be quickly recompiled and hot-loaded over ssh

Excellent tip.

mj-sakellaropoulos, 2021-05-04:
Can confirm, this is working as it should !

hansemro, 2021-05-04:
Polling rate is much slower than downstream, so I am looking into that right now.

hansemro, 2021-05-04:
It is funny how ili212x_touchdata_to_coords has the negation... Maybe there are not many people with ili210x IC to test?

mj-sakellaropoulos, 2021-05-04:
I'm wondering if it's down to firmware or chip variant...

hansemro, 2021-05-04:
active low states?

mj-sakellaropoulos, 2021-05-04:
So then, is this issue resolved? Or do we still need right click ? 😂

hansemro, 2021-05-04:
That doesn't appear to be implemented by the driver...

mj-sakellaropoulos, 2021-05-04:
> That doesn't appear to be implemented by the driver...

Hmmmm....

Also a suggestion : install the xinput_calibrator package for easy calibration

hansemro, 2021-05-06:
> Also a suggestion : install the xinput_calibrator package for easy calibration

Can we use this to calibrate the touchscreen without setting the resolution field?

mj-sakellaropoulos, 2021-05-06:
I tried, but because its invalid from the start, you can't calibrate in a way you would expect.
```
[ Base resolution from driver (square) ]
|
|-->[ Xinput manual calibration matrix ]
```

Since userspace calibration is applied on top of what the input subsystem provides, the values are wrong from the start and you would have to compensate for the specific way they are wrong.

Now there may be a way to force the evdev X_MAX and Y_MAX properties from userspace, but I wouldn't know it.

hansemro, 2021-05-06:
@mj-sakellaropoulos Can you submit a patch to correct the endianness of coordinate data?

The patch I submitted to add the missing negation got approved by the driver maintainers. https://lore.kernel.org/linux-input/a07a1804-28cd-bd07-f3cd-3307be433534@denx.de/T/#ma35a9cf3a843775f5abe8c91ebcb44d453bda3cb

mj-sakellaropoulos, 2021-05-06:
Ok I've seen your patch, I'm looking into it now. (Never done one before)

mj-sakellaropoulos, 2021-05-06:
What branch should I use as my origin for the patch?

hansemro, 2021-05-06:
> Never done one before

Same. GregKH has a nice presentation on this (https://www.youtube.com/watch?v=LLBrBBImJt4).

For email setup/submission, I followed [this](https://nickdesaulniers.github.io/blog/2017/05/16/submitting-your-first-patch-to-the-linux-kernel-and-responding-to-feedback/)

hansemro, 2021-05-06:
> What branch should I use as my origin for the patch? Master ?

I worked on top of v5.12

mj-sakellaropoulos, 2021-05-06:
OK, thanks.

hansemro, 2021-05-06:
Add me to `Tested-by:` tag as I think that is more appropriate.

https://www.kernel.org/doc/html/latest/process/submitting-patches.html#using-reported-by-tested-by-reviewed-by-suggested-by-and-fixes

hansemro, 2021-05-07:
Added patches for touchscreen to main: ee9b12065231151f47774b264e5d590c569114f9

I will leave the issue open until we can find a way to get these patches upstream...

[3 more replies]