hansemro/kc1-linux #38: Add remoteproc support to device tree

hansemro, 2021-05-19:
Resources:
- https://github.com/KFire-Android/kernel_omap_otter-common/blob/945e2d88eda644c5b5d81156b779428987ab4ddf/arch/arm/mach-omap2/board-4430kc1-tablet.c#L973-L983
- `arch/arm/boot/dts/omap4-panda-common.dtsi`

Relevant hardware:
- DSP
- IPU
- Ducati/M3 coprocessor

Relevant kernel config options:
- CONFIG_REMOTEPROC
- CONFIG_OMAP_REMOTEPROC

On downstream kernels, Ducati firmware gets loaded from `/lib/firmware/ducati-m3.512MB.bin`.

`arch/arm/boot/dts/omap4-kc1.dts`:
```
\ {
	...
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;
		...
		dsp_memory_region: dsp-memory@98000000 {
			compatible = "shared-dma-pool";
			reg = <0x98000000 0x800000>;
			reusable;
			status = "okay";
		};

		ipu_memory_region: ipu-memory@98800000 {
			compatible = "shared-dma-pool";
			reg = <0x98800000 0x7000000>;
			reusable;
			status = "okay";
		};
	};
	...
};
...
&dsp {
	status = "okay";
	memory-region = <&dsp_memory_region>;
	ti,timers = <&timer5>;
	ti,watchdog-timers = <&timer6>;
};

&ipu {
	status = "okay";
	memory-region = <&ipu_memory_region>;
	ti,timers = <&timer3>;
	ti,watchdog-timers = <&timer9>, <&timer11>;
};
```

hansemro, 2021-05-19:
<details>
<summary>5.12.0 dmesg</summary>
<pre>
[    0.000000] OF: reserved mem: initialized node dsp-memory@98000000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: initialized node ipu-memory@98800000, compatible id shared-dma-pool
[    5.005432] platform 44000000.ocp:dsp: Adding to iommu group 0
[    9.287597] platform 55020000.ipu: Adding to iommu group 1
[    9.513244] omap-rproc 44000000.ocp:dsp: assigned reserved memory node dsp-memory@98000000
[    9.522674] remoteproc remoteproc0: 44000000.ocp:dsp is available
[    9.536682] omap-rproc 55020000.ipu: assigned reserved memory node ipu-memory@98800000
[    9.550506] remoteproc remoteproc1: 55020000.ipu is available
[    9.602630] remoteproc remoteproc0: Direct firmware load for omap4-dsp-fw.xe64T failed with error -2
[    9.615234] remoteproc remoteproc0: powering up 44000000.ocp:dsp
[    9.633605] remoteproc remoteproc0: Direct firmware load for omap4-dsp-fw.xe64T failed with error -2
[    9.670379] remoteproc remoteproc1: Direct firmware load for omap4-ipu-fw.xem3 failed with error -2
[    9.683807] remoteproc remoteproc1: powering up 55020000.ipu
[    9.984191] remoteproc remoteproc1: Direct firmware load for omap4-ipu-fw.xem3 failed with error -2
</pre>
</details>

hansemro, 2021-05-20:
I think so. https://en.wikipedia.org/wiki/Distributed_Codec_Engine

mj-sakellaropoulos, 2021-05-20:
Afaik it needs custom gstreamer plugin and custom firmware for ducati